frangoteam/FUXA #1000: [FEATURE] Snapping & Alignment

lewsut, 2023-12-07:
Alignment and snapping for the front end.

This would make layouts much simpler

unocelli, 2023-12-08:
Hi,
what do you mean? in which context?

lewsut, 2023-12-08:
sorry I should have had in edit mode of the designer, take Figma as an excellent example, everything snaps in alignment and even suggests equal distances. While this is at the fancy end of the spectrum the likes of Red Lion's crimson have alignment snapping and grid snapping. Currently I have to check the coordinates for true alignment and that's titresome

[image]

[image]