frangoteam/FUXA #999: [FEATURE]Choose device tags support from JSON

gxherror, 2023-12-07:
I get the JSON from ohter platform using WebAPI device, the JSON format is unable to change, I want the tag like this:  temperature->25, version->1.0, but it don't work, how can I do this?
```JSON
{
    "temperature": [
        {
            "ts": 1701939648842,
            "value": "25"
        }
    ],
    "version": [
        {
            "ts": 1701939648842,
            "value": "1.0"
        }
    ]
}
```

my version:FUXA-1.1.15

lewsut, 2023-12-08:
read how to use JSON.parse then use a script in edit project to manipulate the data into the tags you want ...
[image]

gxherror, 2023-12-08:
thanks, i will try this

closed 2024-05-22 by unocelli