frangoteam/FUXA #1003: [BUG] Only one Action can be set, hide/show not possible

lewsut, 2023-12-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Crate shape
2. Click on shape
3. click on Properties 
4. click on actions
5. Select show on x tag ==1
6. click +
7. Select hide on x tag == 0
8. click ok
9. go back into properties / actions, there is only one action so toggle hide show is not possible.

**Screenshots**
If applicable, add screenshots to help explain your problem.

Set details, click ok
[image]

On return...
[image]

**Environment**
 - FUXA version: 1.1.15-1368

unocelli, 2023-12-10:
Hi, I can't confirm the issue, I have just tested it and it works as expected.

lewsut, 2023-12-10:
> Hi, I can't confirm the issue, I have just tested it and it works as expected.

It looks like it's just happening on a few specific shapes, I'll see if I can narrow it down.

closed 2023-12-11 by lewsut