frangoteam/FUXA #1005: [FEATURE] Touch events similar to Mouse down mouse up for touch screen / mobile browsers

lewsut, 2023-12-10:
For example, clicking stuff to see a mouse down up event on a mobile browser is poor.  

How things look on a PC 
[image]

How things look on android
[image]

Miroslav999, 2024-09-09:
This is also relevant for me

unocelli, 2024-09-19:
Hi, This issue should be fixed in the current master branch 2b38e320dfae056308a1ad7a2ab3eff8e13a83ad. let me know if you have any issues.

unocelli, 2024-10-27:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-10-27 by unocelli

Miroslav999, 2025-03-18:
@unocelli , I still have this problem (

unocelli, 2025-03-21:
@Miroslav999 please provide a project where you can reproduce the issue

Miroslav999, 2025-04-04:
@unocelli , sorry, i mean another scenario, open live demo https://frangoteam.github.io/home -> open developer tools (F12) -> tap Toogle device toolbar -> set dimensions any android (Example pixel ) -> tap burger button 
Result: burger button does not respond to pressing