frangoteam/FUXA #1008: [FEATURE] Events and Actions to have AND + OR

lewsut, 2023-12-12:
It would be useful like may other packages to have the ability to AND, and also OR tags to drive events and actions of mimics. 

For instance only show shape_1 if tag x and tag y are true.

unocelli, 2023-12-12:
Hi, You can use scripts that are even more flexible

lewsut, 2023-12-13:
I know, I certainly can, but it's not as fast or as user friendly, that's why nearly all HMI / SCADA applications allow you to do it in the GUI without scripting.

unocelli, 2023-12-13:
The question is: where is the limit, only 2 tags, why not 3? and combined with brackets? and so it gets complicated.
You certainly have an example of how the gui should look right?

lewsut, 2023-12-14:
> The question is: where is the limit, only 2 tags, why not 3? and combined with brackets? and so it gets complicated. You certainly have an example of how the gui should look right?

Well ideally it would be somewhat unlimited, and "scripts" would be accessible at the point of work. Here is an example, obviously this could be extrapolated for any number of tag tests.
[image]

unocelli, 2023-12-17:
Thanks, Yes, indeed the possibility of having the true/false condition via script is a nice feature.