frangoteam/FUXA #1011: Virtual Keypad

robbudge, 2023-12-13:
Testing on   1.1.17-1370
When an Input box has the type 'Number' selected the full Alpha-Numerical keyboard is shown and not the Numerical-Keypad.

Previously the Input Box had a Numerical ONLY switch, this forced a Numerical-Keypad to be shown as the virtual Keyboard.
Its looks like the updated data type drop down is not triggering the Numerical-Keypad on data type Number,

unocelli, 2023-12-13:
Hi, Thanks for the report, I will fix it

robbudge, 2023-12-14:
its actually hit and miss, after a complete restart i now get the correct num-pad most of the times (on existing numerical entry boxes) , but the number order is reversed again, an entry of '2','0' returns 02

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli