frangoteam/FUXA #1020: [FEATURE] sessions and machine safety

mariorl, 2023-12-18:
Safety matters are really hard on these kind of apps (Scada, hmi), since multi view is still a young feature on these systems.

Currently I manage with safety creating a "backplane" visualization/main page, and several buttons to access to cards / dialogs, so that an user must have certain privileges to operate these buttons, and thus gain access to these dialogs. Within these dialogs is where you can open a valve, for example. 

But I can't deal with an user who can login with same user and pass from two devices (or two browser tabs) at same time. It will be a good practice to not allow a second session with same user, since you are accessing something who can moves (a pump, conveyor, etc). In fact, only with this improvement I think the system can get effective safety management. In other systems, the behavior is that an user can login with already logged on credentials on a second device, but then is automatically logged out on the first device.

You can with this scheme made powerful systems with multiple "control/view places", so that in several places there is only autorization to view and then the browser automatically log vith an user account capable only for viweing, and there are also some users accounts with highger user rigths, created to be used in places with control capabilities. In this case the problems that I have encountered over years are that there has to be a mechanism that enables a single device to log in only with an exclusive account, something like a physical location code of the viewing equipment, which makes that only a specific user account can log in to the device. I think this can be achieved with a script that is executed every time a user tries to log in.

Other considerations coud be:
- Set a data field in general view (side / top menu) showing current logged user.
- Page objects does not responding or updating when there are no user loged at all. It will be very useful to give a blank page sowing this issue, a grayed wiev or somethig who tells the operator that visualization / page is no available.

Thanks for your attention, 

Mario JRL.

IonuDragomir1, 2023-12-21:
Hi, @mariorl 

Yes, it'd be great if would exist a data field in general view (side / top menu) showing current logged user.
Normally, when the token expires, the sign in window should be displayed so the user can continue to use fuxa. At the moment, there is a bug when the token expires. No data is longer displayed and the user doesn't know what is happening because no warning is displayed. That's why is best to display the sign in window after token's expiration.

unocelli, 2023-12-29:
Hi, 
@mariorl the issue of security as well as that of access management is quite complicated, I am not sure that blocking multiple sessions is feasible, the closing of a session is not deterministic and there may be a risk that as long as the server is managing the sessions you don't have access.

Sure, I think giving the option to show the current logged in user is a good feature and logout also needs to be handled better, we need to investigate.

mariorl, 2023-12-31:
As a starting point, if the frameworks you are using let you work with sessions, the most important thing to avoid is the posibility of controling something from two devices at same time, it is not exactly concerned to user rigths itself.  

I think most of the work would be done if the "card" type pages had access rights, and if there can't be two accounts logged into that page, even if they are two admin type accounts. Then, it would only be a matter of the system integrator creating general "big" views that can see more than one user account at a time, and "cards" or control views that only allow one logged in user.

Other hmi platforms give the ability of perform user login/logout (as button actions or via scripting), to the system integrator; but Fuxa is a multiview system, not an simply hmi.