frangoteam/FUXA #1021: [FEATURE] Port change

Bundyra, 2023-12-19:
Chaning default port from 1881 to custom port. Currently in settings we can only see port on which server is working, but we cannot change it. It would be nice to have that option, so we admin doesn't have to change port manually in files.

mauroalexandre, 2023-12-20:
Hi.

Look this line https://github.com/frangoteam/FUXA/blob/1467713dfdd30299fc4bd34ef045804c22133f44/server/settings.default.js#L10

First, FUXA tries to read the environment variable PORT; if it does not exist, it uses the value 1881.

You can change the default port setting your enviroment var PORT, for example to use port 9090:

```
[upmauro@archlinux server]$ export PORT=9090
[upmauro@archlinux server]$ npm start
&gt; fuxa-server@1.1.15-1376 start
&gt; node main.js
2023-12-20T17:31:59.075Z [INF] 	FUXA started!
2023-12-20T17:31:59.077Z [INF] 	WebServer is running http://127.0.0.1:9090/
```

Bundyra, 2023-12-21:
But what if i want to run two fuxa servers on the same machine?

Bundyra, 2023-12-21:
It doesn't work for me either. Any ideas why?
[image]

Bundyra, 2023-12-21:
It's not working for me. Any ideas why?
[image]

unocelli, 2023-12-21:
setting.js is the default setting, but changes from UI are saved in mysettings.json and overwrite the values of settings.js (look uiPort)

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli

Bundyra, 2024-01-08:
Im still a bit confused.. 
My problem still occurs.
My scenario:
Environment variable PORT=1884
[image]
I made some changes in client dir, for example changed favicon and title in index.html, so i did ng build.
I start server once again:
[image]
[image]
[image]

Bundyra, 2024-01-08:
Im still a bit confused..
My problem still occurs.
My scenario:
I changed environment variable PORT=1884.
[image]
Server starts, fuxa is loading.
Next, i do some changes in the client dir, for example I change title of the site in index.html and change favicon.
I do ng build.
Start server again:
[image]
and then:
[image]
[image]
What am I doing wrong?