frangoteam/FUXA #1022: [FEATURE]

2697403929, 2023-12-20:
Hello, after I imported the svg and configured it as shown in the picture, why only the area I circled in the picture is flashing? Normally the entire object should be flashing. I tried many, either a small part of it would flash, or it wouldn't flash at all. Why is this happening?
[image]
[image]
[image]

mauroalexandre, 2023-12-20:
Can you attach SVG?

2697403929, 2023-12-21:
[image]
[image]
This is the original file

2697403929, 2023-12-21:
[image]
[image]

unocelli, 2023-12-29:
Hi, 
these images consist of more than 1000 elements, I honestly find it a challenge to try to select one or a group of them to edit, for your goal I would add a new element on over.

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli