frangoteam/FUXA #1023: [BUG] Dialog input type mouse move inside card

mauroalexandre, 2023-12-20:
**Describe the bug**
If dialog Mode of input field is "Dialog" when you try use inside a card, the card follow the mouse when input is clicked.

**To Reproduce**
1 - Edit project and set layout dialog mode to Dialog.
2 - Create a view with input number
3 - Open this view from another view using action open card.
4 - Click on input field. 
5 - Move your cursor

**Environment**
 - FUXA version: latest (v1.1.15)
 
 Attached video example.

https://github.com/frangoteam/FUXA/assets/68612845/78b3925b-853f-4227-bd1f-574dcceb2fad

unocelli, 2023-12-20:
Hi Mauro, Thanks for your report.
I think it would be correct to close the input dialogue if the click occurs outside the input area.

mauroalexandre, 2023-12-21:
Hello Unocelli,

However, as of today, the card (popup) follows the mouse cursor. Is this the expected behavior?

unocelli, 2023-12-21:
Yes right, I think the bug is that the Card should only move with the mouse pointer on top of Card
[image]

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli