frangoteam/FUXA #1024: [FEATURE] Users with view roles can open panels

mauroalexandre, 2023-12-21:
**Describe the feature**
Sometimes there are pieces of information contained within panels. May I submit a pull request to enable users with the viewer role to interact with these elements through click events (such as opening a panel, window, or card)?

**Describe the solution you'd like**
If the element has a click event, such as opening a panel, window, or card, then a user with the viewer role is able to interact with it.

**Additional context**
For instance, consider a view featuring a 'motor engine.' All data pertaining to this engine is displayed in a panel that opens when a user clicks on the element. Currently, if a user has only the viewer role, they are unable to access this information.

unocelli, 2023-12-30:
You can try to play with authorisation in Property of Controls and Shapes to configure the access to the View
[image]

the View authorisation settings are missing

unocelli, 2025-01-01:
I’m going to close this as resolved. let me know if you have any issues.

closed 2025-01-01 by unocelli