frangoteam/FUXA #1025: [FEATURE]Can variable links be preserved when copying elements or cloning views?

pystudy123, 2023-12-22:
**Describe the feature**
The requirement is, for example, I made an indicator light with many colors that change according to various variables. Then if I copy this indicator light to the current or other views, the indicator light will return to its original appearance, and the color changes based on the data will disappear, which is very inconvenient.

**Describe the solution you'd like**
When pasting, can I choose whether to keep the link or keep the function?

**Additional context**
[image]

pystudy123, 2023-12-22:
Or maybe I have done many effects in one attempt, and after cloning a similar attempt, I have to do it again.

unocelli, 2023-12-22:
I suppose you have try with Shapes of General section. For all other Shapes or Controls the property should be copied too.
There is already a issue open #1015

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli