frangoteam/FUXA #1026: [FEATURE]Can I add a tab page to the view?

pystudy123, 2023-12-22:
**Describe the feature**
That is, in a view page, I can click the button to display different content

**Describe the solution you'd like**
It is worth not clicking the button to display different view pages. For example, if there are too many view pages, reaching dozens, I would like to think of some, such as several attempts, through an effect similar to a tab page, to achieve it in one view.

**Additional context**
[image]

unocelli, 2023-12-25:
Hi, I suppose it is the same feature discussed in #925, it is coming soon.

unocelli, 2024-03-17:
is available in the latest version,  let me know if you have any issues.

closed 2024-03-17 by unocelli