frangoteam/FUXA #1032: [QUESTION] Can I change TAG VALUE from internal device using script?

mauroalexandre, 2023-12-26:
**Describe the feature**
One simple example, I want do create a clock in my view (to show current date/time). My first is described here (), but, I was curious, is possible a change value for tag using server or front script?

lewsut, 2023-12-27:
Like this?

Create a script to write to a string tag the date and time every second...
[image]

then display that tag...

[image]

edit script for the time format you want....

[image]

mauroalexandre, 2024-01-02:
Wow @lewsut thanks for yout help. Now I see "functions" in editor. My fault.

Thanks alot.

closed 2024-01-02 by mauroalexandre

Jue07, 2024-02-22:
> Like this?
>
> Create a script to write to a string tag the date and time every second... [...]

Hi, the date time script does not appear in the tag selection. How to configure it?