frangoteam/FUXA #1034: History table shows multiple data issues

zhuxianghuan, 2023-12-28:
After a device tag I configured went offline, influxdb stopped storing it. The historical chart displayed normally, but the historical table control displayed abnormally.
[image]
[image]

I think column 4 in the last picture should be null.
It is normal when I configure only this offline tag in the history table.
[image]

unocelli, 2023-12-30:
Hi, Thanks for the report.
What do you mean with offline tag? and what kind of connection do you have?

zhuxianghuan, 2024-01-16:
Hi, I am using mqtt connection, updating the tag topic every second, and going offline when it stops sending.

unocelli, 2024-03-17:
Hi, sorry for the delay.
I suppose the problem is that mqtt runs per subscription and fuxa assumes that your value has not changed since no new value is sent. i think we need a specific handling with a timeout if the value is not updated