frangoteam/FUXA #1047: [BUG] Clunky keypad entry, Decimal Point key causes number deletion.

lewsut, 2024-01-11:
Generally the selection of numbers, is not so intuitive, personally I would expect to type a new number in after I click the box not add to what is there. However that could just be preference. 

Then the decimal point key seems to cause deletion of the entry.

Version 1.1.17-1399

[image]

robbudge, 2024-01-15:
I did notice the same, but was only on Safari not chrome.

lewsut, 2024-01-16:
Mine was on chrome and firefox

unocelli, 2024-01-18:
Hi, the Virtual Keyboard with input type number have some issue, I think that I have found a compromise in the last commit.

unocelli, 2024-03-17:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-03-17 by unocelli