frangoteam/FUXA #1048: Port change problem still occurs.

Bundyra, 2024-01-12:
Im still a bit confused..
My problem still occurs.
My scenario:
I changed environment variable PORT=1884.
[image]
Server starts, fuxa is loading.
Next, i do some changes in the client dir, for example I change title of the site in index.html and change favicon.
I do ng build.
Start server again:
[image]
and then:
[image]
[image]
What am I doing wrong?

_Originally posted by @Bundyra in https://github.com/frangoteam/FUXA/issues/1021#issuecomment-1880554295_

unocelli, 2024-01-12:
Hi, I can only confirm in FUXA it works as expected! so the problem lies in your modifications, I would look there

Bundyra, 2024-01-12:
I just reproduced it without making any changes in the files, so i wouldn't be so sure about the problem with my modifications. All i did was npm install in the server directory, then npm install and ng build in the client directory.
I start fuxa: 
[image]
And then:
[image]
But if you are sure that everything is working as expected then ok.
Thank you for your answer.

unocelli, 2024-01-12:
try 
`ng build --configuration=production`

Bundyra, 2024-01-12:
Works perfect. Thank you for your help!

unocelli, 2024-03-17:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-03-17 by unocelli