frangoteam/FUXA #1050: [BUG] Date and Time option in Input control doesn't work correctly

KamilGrabowski01, 2024-01-15:
**Describe the bug**
While attempting to insert data into the input control, using both date/time and separate date and time input types, two windows appear simultaneously. One displays the date/time, while the other presents a numeric keyboard. It is not possible to save the date in the first window; clicking elsewhere prevents the date from being saved, and the window closes, leaving only the numeric keyboard window. If you attempt to input data using the numeric keyboard, it does not function as expected, as it always leave the default data, no matter what you click. The only option for data insertion is to utilize the first window to input date/time and subsequently save it by pressing 'Enter' in the second window.

**To Reproduce**
Steps to reproduce the behavior:
1. Create input control with date/time or separate date and time input types.
2. Click on it to insert data.
3. Notice that two windows have popped up simultaneously—one with date/time and the other with a numeric keyboard.
4. Try to insert any date/time.
5. See that any attempt to save data using only one of the windows is not possible.

**Expected behavior**
In the date/time selection window, there should be an 'Enter' button that saves the date/time to a variable, and the window with the numeric keyboard should not open.

**Screenshots**
Creating input control.
[image]
Trying to insert data, see that two windows appear simultaneously.
[image]
After choosing the date, the second window with the numeric keyboard remains open.
[image]
If we click enter, it will save the choosen date.
[image]
But if we click any other button, it will reset the date.
[image]

**Environment**
 - FUXA version: 1.1.17 - 1399
 - Node.js version: 16.20.2
 - npm version: 8.19.4
 - Platform/OS: Windows 10
 - Browser: Opera

unocelli, 2024-01-20:
Hi, Thanks for the report.
let me try to fix it

unocelli, 2024-03-17:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-03-17 by unocelli