frangoteam/FUXA #1051: Api get project: RangeError: Invalid string length

Bundyra, 2024-01-15:
Hello, i have a problem with one of my projects.
After enabling token authentication and logging out, page is not loading.
Logs from console:
[image]
[image]
Any idea what can cause this?
It happens only in certain project.
Loading this project on fresh fuxa isn't fixing it either.

mauroalexandre, 2024-01-23:
Can you reproduce it in any project that you can send us? Is the one that gives the error a project that we can access?

unocelli, 2024-08-23:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.