frangoteam/FUXA #1052: [FEATURE] avoid unsafe accesses

mariorl, 2024-01-17:
Hi, I created a simple system for safe use on several sites (browsers) at same time, using external python ans some js scripts within Fuxa. 

[image]

This is a base project for a multipoint system. There will be no buttons to change from one view to another, which helps avoid many problems, such as controlling something from two places at the same time.

However, there are two ways to bypass this access system, as far as I know:

    Writing "/editor" after the Fuxa server address, since I'm not using Fuxa Authentication. I think this issue could be easily overcome by enabling Fuxa authentication "only for the editor," as suggested in the settings button. However, I observe that Fuxa token expiration affects the other views as well. In fact, it works in the opposite way I expected; the expiration affects user views instead of affecting the editor view.

    Accessing via "http://localhost:1881/view?name=[the View name]". Perhaps, in the page settings, a switch could be included to avoid this kind of access, or even better, allowing a script to run on load to determine if it is allowed (but in that case, the script needs to know how the view is being accessed).

Kind regards,
Mario JRL.

unocelli, 2024-01-21:
Hi, if you enable "only for the editor" all the Views should work as expected and the token expiration don't have any affects to the Views

unocelli, 2024-08-23:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-08-23 by unocelli