frangoteam/FUXA #1055: MCPROTOCOL (tested in Mitsubishi system Q)

mariorl, 2024-01-19:
Hi,
I added McProtocol js to my Fuxa app using scripts and it works reasonably well. If someone wants a copy of the app please tell me. But, in a lack of api for write user protocols, it will be useful a kind of timer faster than a second, because the polling protocol part is as fast as scripts are called, and the max frecuency I observed is every second. (writing is faster, almost instantaneous).

Kind regards, Mario JRL.

unocelli, 2024-01-23:
Hi,
interesting, is this something that could be added in the FUXA connections?

mariorl, 2024-01-23:
I don't know, since I am currently using JavaScript at a beginner level. I used the Fuxa scripting feature to create several scripts, one for general polling and another for group adressing and for writing. If you have an API or something similar with functions designed for wrapping other PLC protocols I think it should be easy to establish a new Fuxa protocol. MCprotocol.js itself has several bugs, that I had to address, but it primarily involved terminating the communication task in case of a timeout and then restarting it. Despite these issues, I have noticed that a good stability is achieved this way; communication, at least for writing, is very fast, and pollig is a bit slow but I dont get crashes.

sephi251090, 2024-06-25:
This feature intrest me too :)

mariorl, 2025-04-01:
> Hi, interesting, is this something that could be added in the FUXA connections?

It probably wouldn’t be complicated to add MC Protocol to FUXA connections, but the question is broader: it would make FUXA accessible to many more people who currently can't use it due to the lack of a driver for their devices.

I know today OPC UA is available for main trademarks of processosrs, but still there are thousands of old devices in today plants who cant be accesed in this way.

The question itself is about the possibility of having an API or a guide for developing new protocols, which, of course, could be integrated into FUXA connections.

henjoe, 2025-04-07:
> It probably wouldn’t be complicated to add MC Protocol to FUXA connections, but the question is broader: it would make FUXA accessible to many more people who currently can't use it due to the lack [...]

What do you mean by "developing new protocols", like creating new Device connections in the Fuxa connections? Like added supported protocols?
If yes, there is actually a branch for this.

jingshui127, 2025-04-08:
我现在是将三菱协议转化为MQTT，然后MQTT与FUXA对接。目前运行良好