frangoteam/FUXA #1056: Press and hold on touch screen devices

IonuDragomir1, 2024-01-19:
Hi,

I'm running FUXA on a device with touch screen and Windows 10 OS. I'm facing the following problem as you can see in [my video](https://gemoo.com/tools/upload-video/share/606955599585480704?codeId=PYadWqkp2BqpJ&card=606955594510372864).
On the circle is a toggle bit function. If bit's value is true, circle's color is red. Otherwise, circle's color is gray.  If I use the left mouse click on touch screen device, it works always as expected. If I use the finger on touch screen device, two scenarios arise:
1) When I hold the finger a little bit longer on the screen, appears the menu corresponding for right click action. This right-click menu shouldn't be appearing if I hold the finger on touch screen for any time period. It happens even though I adjust the touch's sensitivity. I have disabled from control panel (on Pen and Touch section) the option "Enable press and hold from right-clicking".  This option has no effect upon the web browser. 
2) When shortly taping the circle, the toggle function doesn't execute always.

robsori, 2024-01-19:
Yes, I also confirm this problem. On any Touch Screen, when you press a button, the menu dedicated to the right mouse button appears and becomes annoying for operation. I was not able to remove this menu from the windows operating system settings. Maybe there should be a configurable option in Fuxa to remove this menu for working with the Touch Screen.

unocelli, 2024-01-25:
Hi, are you sure it is not a setting on the OS? it seems to me that you can also set the hold delay.

IonuDragomir1, 2024-01-26:
Hi,
This option is available only on touch screen devices.  I have disabled from control panel the hold setting. It has effect only on Windows OS, but not on browser. So, as @robsori wrote, it should be a configurable option in Fuxa to remove this menu but only on runtime.

unocelli, 2024-09-19:
Hi, should be fixed in recent commits, please try with current master branch

unocelli, 2024-10-27:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-10-27 by unocelli