frangoteam/FUXA #1058: [FEATURE] Ability to set/toggle bits with event

GrantC27, 2024-01-19:
**Describe the feature**
Adding the ability to set or toggle a specific bit with an event such as mouse click

**Describe the solution you'd like**
Either adding bit mask to the set value option in events, or adding new options to the event to toggle and set specific bits

**Additional context**
When trying to set a specific bit in a word, I can use the switch object linked to a tag and a bitmask, with the value corresponding to that bit to turn that bit on and off. I would like a similar functionality on other objects. To achieve this, we could use the mouse click event

Github-User-002, 2024-03-21:
I use a Vacon to control a 22kW motor and the start / stop function I use,  uses binary coded input, so when I want to start the motor reversed/forward with ramping and coasting I want to be able to send this binary coded. Now I need to use a script to do this...

_Also the toggle function could be one field for toggle on and other for toggle off. trinary toggling is also useful for Start/Hold/Stop or the same 3 states as Stop light for cars, ordering could be 123 123 or 12321...
3 pole switch could also be useful_

unocelli, 2024-11-20:
I'm working on this feature

unocelli, 2024-11-24:
Available in the master branch

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-12-11 by unocelli