frangoteam/FUXA #1060: [Question]  mapping floating point value in modbus

henjoe, 2024-01-22:
Hi,

I have a modbus device (which is a power meter) with holding register 6-7 (two holding registers) being its voltage value.
it is a floating point value as per their documentation. 
My question is how can I map the data in the Fuxa? I am using modbus RTU for this.

is my configuration below is correct?
[image]

and also, what is the meaning of divisor? 
Thanks!

Regards,

IonuDragomir1, 2024-01-23:
I have encountered the same issue. More details at [#682](https://github.com/frangoteam/FUXA/issues/682)

GrantC27, 2024-01-24:
I successfully read 32 bit floats from a Schneider PLC and an ABB PLC. The type I use is Float32MLE. If I look at the value in Modbus poll I use little endian swap byte. Seems to work well for me - I'm sure RTU has the same config. The register address is 1 more than what is stated in the PLC. For example, the value in the below screenshot is actually at %MW514

[image]

unocelli, 2024-03-17:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-03-17 by unocelli