frangoteam/FUXA #1063: [BUG] Won't accept time update from mobile

lewsut, 2024-01-26:
Works on PC browser, however mobile (android) time interface fails to take / update values. 

https://github.com/frangoteam/FUXA/assets/25065897/12f89312-6116-4547-9beb-6763185dcb85

unocelli, 2024-01-28:
Hi, Thanks for the report
in master branch you should find a fix for this issue. new the datetime input control have a button to confirm.
[image]

lewsut, 2024-01-28:
Excellent, I'll check it out if I can, I'm using docker and I seem to be only able to get Version 1.1.17-1399 and nothing newer currently, I see their are more recent commits but have no idea what controls the updates as when I rebuild my docker it should grab the newest commit but it doesn't...

I have used latest, 1.1.17 and RC I keep getting the 1.1.17-1399 (on latest I actually get 1.1.17-1398!).

lewsut, 2024-01-28:
Checked out the new version, have you tried this on a mobile browser? The tick sits under the time text so it looks very odd no matter what the alignment is , and clicking it does nothing, the result is exactly the same as in the video above.

unocelli, 2024-01-31:
I suppose that is not the only problem with mobile browser compatibility.

lewsut, 2024-02-01:
As this shows, it doesn't work with "Enabled update" in a regular NON mobile browser. 

[image]

unocelli, 2024-02-04:
with enabled update should be fixed, Thanks!

closed 2026-11-13 by unocelli