frangoteam/FUXA #1065: [Question] Create custom Tag calculation

henjoe, 2024-01-28:
Hi, I just dont know if this is possible in Fuxa, simple create a tag that its value is based from other tags.
For example: 
TagA = (TagB * TagC) / TagD

I know we can create this thru scripts, however scripts are just run based on events, in the concept of SCADA, tag calculations are event (when pressed by button or etc) or time driven (every period of time).
What I wanna achieve is everytime there is a change value on my calculations, it automatically calculate the output. 

Regards,
Henjoe

lewsut, 2024-01-28:
Scripts can be run cycle also not just events. Have you tried that, that's as quick as you are going to spot a change. 

Like in the script you would have  if ( tagA != tagALast) { do something  else tagALast = tagA;}

henjoe, 2024-02-01:
> Creating Tag value changed event to start a script would be a good feature

Yes that is an awesome idea, and better put it in a Fuxa server adding tags options so we can use all tags connected to it regardless of the type (opcua, modbus and etc). 

Simply add another property of a Tag class to have a flag if it is used in the calculations 
like 
if tag.used = true then its value will be put in some variable (varsValueChange)
Then this variable will be emitted _emitValues(varsValueChange) to be checked by the Fuxa Server (Checking of change will be based on how fast the polling of the fuxa server)

unocelli, 2024-08-23:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-08-23 by unocelli