frangoteam/FUXA #1066: [BUG] Text display with numbers doesn't work as expected

lewsut, 2024-01-28:
Annoyed by issue #1063 I thought until it's fixed I would use a walk around of a text input and interpret the string to get my hh:mm values. Unfortunately this doesn't work as expected either, I get my falues just fine back to however the frontend fails to display numbers followed by a non number correctly.

Setup:

[image]

[image]

Entering my string 

[image]

press enter...

[image]

This is a text string I expect to see 09:55 

Back to the drawing board, I'll use two separate number input boxes for now, however it's impossible to display a leading 0 in an input box so it doesn't look good.

unocelli, 2024-01-28:
Hi, thanks for the report. I will fix it as soon as possible. however feel you free to do a PR with the fix.

lewsut, 2024-01-28:
> Hi, thanks for the report. I will fix it as soon as possible. however feel you free to do a PR with the fix.

I wish I had the know how to help fix these issues myself! Thanks for your work man, I know I report a lot of bugs!!!

My knowledge is c/c++ and PLCs, so I'm a bit useless, I wish I could help more!

lewsut, 2024-02-01:
Same issue, different place 

[image]

unocelli, 2024-02-04:
with last commit should be fixed, Thanks!

unocelli, 2024-03-17:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-03-17 by unocelli