frangoteam/FUXA #1067: [BUG] move tools with arrows

Tsigh, 2024-01-29:
**Describe the bug**
When moving chosen items text and figure moves in different pace than output value/LED gauge.

**To Reproduce**
Steps to reproduce the behavior:
Make a block and chose everything in it and the figure itself.

**Expected behavior**
A clear and concise description of what you expected to happen.
Move the block with the arrows left, right, up and down. 
Left and up only moves LD and output value.
Right and down has different paces for text/figure and LED/output value.

**Screenshots**

This is going right, then left the same number of steps.

[image]

[image]

[image]

**Environment**
 - FUXA version: 1.114-1243
 - Node.js version: -
 - npm version: -
 - Platform/OS: Windows
 - Browser: FireFox

unocelli, 2024-01-31:
Hi, with the current version 1.1.17 I cannot reproduce it, some problems with the editor have been fixed

Tsigh, 2024-02-02:
Now in FUXA  1.1.17 -1399

The text follows along, but the buttons (and LED as well) doesn't

[image]
chosing everything.
[image]
moving down 10 steps by arrow down.
[image]
then regretting and going up 10 steps by arrow up.
[image]

unocelli, 2024-02-04:
I wonder if it is a browser problem, did you try to reproduce it with the chrome browser?

Tsigh, 2024-02-06:
Yes indeed, Chrome solves the problem! Thanks for the help. FUXA gets
better and better the more I use it!

closed 2024-02-06 by Tsigh