frangoteam/FUXA #1072: [FEATURE] Call alarm control on a view

IonuDragomir1, 2024-01-31:
Hi,

I have seen that now, on a MVC's container can be added the alarms. A very useful feature would be to call the alarms/history alarms control on a simple view. The need comes for observing the alarms of certain mechanism. For example, when the user clicks on pump P1, a view with it's alarms should be opening. The easiest way I see at first thought would be to filter through database for alarms which have group's name "Pump1".

unocelli, 2024-02-04:
Hi,
the idea would be to add to the table control, the possibility of having the data from a query. As UI a json editor to configure the table properties.

IonuDragomir1, 2024-02-05:
Hello,
So the table control which would contain only the alarms of a certain mechanism should be colored according to mechanism's configured alarms. Please see an example of how the table should be looking like.

[image]

unocelli, 2024-10-23:
Hi, is available in the latest version (Table control), let me know if you have any issues.

closed 2024-10-27 by unocelli