frangoteam/FUXA #1074: [BUG]

Welsyntoffie, 2024-02-03:
Properties dialog of shape does not retain settings.

**To Reproduce**
Steps to reproduce the behavior:
1. Insert shape
2. Select properties
3. Change Min Max value to Boolean (True[1-1]) - Click OK
4. No go to properties again and the previous setting reverts. Clicking OK saves the reverted settings.

**Expected behavior**
To retain the setting for Range or Boolean

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - FUXA version: 1.0.0 (Home assistant AddOn)
 - Node.js version: 
 - npm version:
 - Platform/OS: Home assistant OS
 - Browser: Firefox, Chrome and Brave

MatthewReed303, 2024-02-04:
Hi @Welsyntoffie a lot of fixes have been done since v1.0.0, try the latest under releases, try via docker. The latest is now v1.1.17

Welsyntoffie, 2024-02-04:
Was using the home assistant addon, most likely the addon version I checked. I did how ever setup a docker container for the latest version. According to about version is 1.1.18-1415.
[image]
This is what I select and save then ok.

[image]
This is what I get when I want to edit again.
Which is fine, because it still works as expected, but was really confusing in the beginning. I have a sensor that is True/False that did not work. I thought this was the issue. Turns out I had to convert the True/False to 1/0 via a template.

unocelli, 2024-02-04:
I understand that it can be confusing, it is a matter of economising the saving of the display option selected for each property

Welsyntoffie, 2024-02-05:
Thanks, I'll close this, as it's not a real issue or a bug then.

closed 2024-02-05 by Welsyntoffie