frangoteam/FUXA #1075: [Question] Apply property changes directly in the JSON file.

mauroalexandre, 2024-02-03:
I developed a tool to apply style (color, borders, etc.) to all elements of the project at once.

I do this through a regular expression applied to the project's JSON file.

However, something strange happens. When I apply it and reload the project, the style doesn't change. I need to go into each element and save the properties, then they apply.

I noticed that there is a field in the JSON called svgData.

Would I need to change it there too? Is there a simple way to 'recompile the whole project' and generate the jsonData again?

unocelli, 2024-02-04:
Hi,
I am not sure if I understand what you mean.

mauroalexandre, 2024-02-04:
@unocelli I'm sorry, I'll try to be clearer in my question.

**Steps to Reproduce:**

1. Create a blank project.
2. Add 1 pipe to the project.
3. Save the project as JSON.
4. Open the JSON file in a text editor.
5. Modify a property of an SVG element (for example, change the pipe's color).
6. Save the file.
7. Load the project again.
8. The color change is not visible; open the pipe's property through the editor.
9. Do nothing, just click OK.
10. The color change should now be applied.

Please refer to the attached video for visual demonstration.

https://github.com/frangoteam/FUXA/assets/68612845/19232764-8f3e-4556-a8d3-103c6060b560

unocelli, 2024-02-08:
Hi,
you have to search in svgcontent, a pipe is a svg element composed from more 'path', so from where you changed the color you take the id and search (it should be "id": "PIE_...") when you find the svg element which should be a 'g' you can change the fill and stroke attributes of the 'path' element inside the 'g' element.

mauroalexandre, 2024-02-08:
Understood, but I would like to change several other things. Is this svgcontent tag generated by reading other properties from the JSON, right? At what point is the svgdata generated?"

unocelli, 2024-03-17:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-03-17 by unocelli