frangoteam/FUXA #1077: Report not generating with tag value,

shyamc2u, 2024-02-06:
Report not generating on schedule time with tag values,  
time stamp generating as i set to 5mins, but tag values at that time is not generated in reports, 
tag value live data is displayed on the Modbus tags and SCADA pages is normal.
i checked the sqlite db, tag values are available. 

**Please help me what to do for report generation with tag values.** 
_**kindly give me solution**_ 

and alarms report is generating is good.

unocelli, 2024-02-08:
Hi,
I just tested and for me is working as expected. can you send me the project?

shyamc2u, 2024-02-09:
[fuxa-project.json](https://github.com/frangoteam/FUXA/files/14216142/fuxa-project.json)
I am restarting the PC (server start time is 9 am to 5 pm), fuxa running well but reports with tag values did not appear in the 
pdf 
file.

Jue07, 2024-03-15:
Hi..I am also facing the same issue..can you help me?

shyamc2u, 2024-04-06:
It's working fine, but don't restart PC , don't shutdown PC and don't close the fuxa application, for 24x7 365 days. If you did anything reports not generate for 24 hrs. Then again next  day on words it's working fine.

unocelli, 2024-05-17:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-05-17 by unocelli