frangoteam/FUXA #1081: $getTagId('Tag name');[FEATURE]

mariorl, 2024-02-15:
**Describe the feature**
When you create several controlers, for example several siemens S71500 devices, and there are the same tags names inside them, a problem is raised with $getTagId('Tag name');, because the device is not an argument for the function. This could be addressed giving different tags names for vars inside every device. I didn't test the scope of this issue in other functions who works with IDs.

**Describe the solution you'd like**
May be giving a "device" argument to that function -> $getTagId('Tag name', 'device');

MatthewReed303, 2024-02-15:
I ran into this issue also trying to use scripting and tag names. I have remote pump stations with OPC-UA and each device is programmed the same with same tag names, I ended up having to run a separate instance of Fuxa on Docker for each pump and just use different ports. Then I had a master Fuxa that opened the other Fuxa pages in an IFrame, it's rough, but is a work around in the meantime. I wasn't going to change tag names on the PLC end, they need to stay the same. I agree having an option to define the device is defiantly needed.

unocelli, 2024-03-17:
is available in the latest version,  let me know if you have any issues.

closed 2024-03-17 by unocelli