frangoteam/FUXA #1082: [BUG] TypeError: Cannot read properties of undefined (reading 'layout')

IonuDragomir1, 2024-02-15:
Hi,
I get on browser's console the error:  

> TypeError: Cannot read properties of undefined (reading 'layout')

This error is encountered  when I add an input field on a screen which is called on panel control. 

**Screenshots**
[image]

**Environment**
 - FUXA version: 1.1.18 - 1421
 - Node.js version: v14.21.3
 - npm version: 6.14.18
 - Platform/OS: WIN 10
 - Browser: Chrome

unocelli, 2024-02-15:
Hi, thanks for the report.
can you please attach the project so that we can reproduce it?

IonuDragomir1, 2024-02-16:
[fuxa-project_InputField.json](https://github.com/frangoteam/FUXA/files/14307740/fuxa-project_InputField.json)
To get the error, press on E1 button to open view E1 in panel. 
I didn't test for all fuxa's controls. The error might appear also on others.

unocelli, 2024-02-16:
Hi, thanks for the project...I can't reproduce the exception. please try with the last one version

IonuDragomir1, 2024-02-19:
Hi, I have used the last version of FUXA (Version 1.1.18-1421). It's not this the last?

unocelli, 2024-02-23:
1.1.18-1431

IonuDragomir1, 2024-02-26:
Yes. With this version, I don't get the error.

unocelli, 2024-03-17:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-03-17 by unocelli