frangoteam/FUXA #1083: [BUG] TypeError: Cannot read properties of undefined (reading 'views')

IonuDragomir1, 2024-02-15:
Hi,
I get on browser's console the error:

> TypeError: Cannot read properties of undefined (reading 'views')

I get this error when trying to open a page, card or dialog from a view which is called on a panel control.
The Open iframe and Open Window actions are working properly when are  fired from a view which is called on a panel control. 

**Screenshots**
[image]

**Environment**
 - FUXA version: 1.1.18 - 1421
 - Node.js version: v14.21.3
 - npm version: 6.14.18
 - Platform/OS: WIN 10
 - Browser: Chrome

unocelli, 2024-02-15:
Hi, thanks for the report.
can you please attach the project so that we can reproduce it?

IonuDragomir1, 2024-02-16:
[fuxa-project_Open.json](https://github.com/frangoteam/FUXA/files/14307721/fuxa-project_Open.json)

unocelli, 2024-02-16:
Hi, thanks for the project...I can't reproduce the exception. please try with the last one version

unocelli, 2024-02-23:
1.1.18-1431

IonuDragomir1, 2024-02-26:
Even though I don't get the error anymore, the  open  page, card or dialog are not working from a view which is called on a panel control.

unocelli, 2024-03-17:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-03-17 by unocelli