frangoteam/FUXA #1084: [BUG] and [FEATURE] on control's property settings

IonuDragomir1, 2024-02-16:
Hi,

**BUG:**
A few days ago I was testing, on FUXA, button's enable authorization property. When logged as Operator, I was able to press on a button which was enabled only for Engineer. Even though button's action wasn't executed, the operator was able to press it. Normally, the button should have been  completely disabled and  when the user is logged as operator and button enabled for engineer.

**FEATURE:**
All SCADA environments have the possibility of disabling controls (buttons, input fields, slider, switch, select value, etc.). This option should be available independent of the type of logged user (admin, operator, engineer, etc.). There are many practical scenarios when a control must be disabled for a period of time until it can be enabled to accomplish its function.
So would be useful if on control's property page would be added a check box to disable/enable it.

[image]

unocelli, 2024-02-17:
Hi, your request already exists and should be applied to all Shapes and Controls #339
why do I have to show a button that I cannot use?

IonuDragomir1, 2024-02-19:
Hi, 

> your request already exists and should be applied to all Shapes and Controls https://github.com/frangoteam/FUXA/issues/339

Sorry, I forgot about the request I made almost two years ago. 😅 

> why do I have to show a button that I cannot use?

Let's say you have a window with four buttons from which you can control a mechanism.

[image]

So, when the mechanism is on AUTO mode, the buttons OPEN and CLOSE should be disabled. When the mechanism is on MAN mode, the buttons OPEN and CLOSE should be enabled.

unocelli, 2024-02-23:
OK, but then it is not a question of authorizations. 
It could be solved by adding the function via script.

kheinrich188, 2024-04-17:
> So, when the mechanism is on AUTO mode, the buttons OPEN and CLOSE should be disabled. When the mechanism is on MAN mode, the buttons OPEN and CLOSE should be enabled.

Hi, that would be exactly what I need.

kheinrich188, 2024-04-17:
What would also be practical to avoid duplicate shapes is that disabled works on shapes and then no actions can be carried out. I currently have 2 shapes that are either hide or show depending on which mode is active, since one of the shapes is also clickable

jay-jjwu, 2024-08-05:
i also need this feature, is it resolved?

kheinrich188, 2024-08-06:
@jay-jjwu not yet