frangoteam/FUXA #1085: Don't let me install fuxa on raspberry

Brandix97, 2024-02-17:
Hello I tried to install FUXA on raspberry and show me this error:

```
npm WARN notsup Not compatible with your version of node/npm: triple-beam@1.4.1

npm ERR! code EACCES
npm ERR! syscall access
npm ERR! path /usr/lib/node_modules
npm ERR! errno -13
npm ERR! Error: EACCES: permission denied, access '/usr/lib/node_modules'
npm ERR!  [Error: EACCES: permission denied, access '/usr/lib/node_modules'] {
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/usr/lib/node_modules'
npm ERR! }
npm ERR!
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR!
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.
```
I have installed raspbian 64 bits and I use this commands:

```
sudo apt-get update
sudo apt-get upgrade
curl -sL https://deb.nodesource.com/setup_12.x | sudo bash -
sudo apt-get install nodejs 
npm install -g --unsafe-perm @frangoteam/fuxa
```

unocelli, 2024-02-18:
Hi, I will try to use nodejs 14.21.3

Brandix97, 2024-02-18:
Hello, if someone have the same error than me to solve it make this:

```
sudo apt-get update
sudo apt-get upgrade
curl -sL https://deb.nodesource.com/setup_12.x | sudo bash -
sudo apt-get install nodejs 
npm install -g --unsafe-perm @frangoteam/fuxa
sudo npm install -g --unsafe-perm @frangoteam/fuxa
sudo fuxa start
```

closed 2024-02-18 by Brandix97

cyclone11, 2024-09-13:
> 您好，如果有人遇到与我相同的错误，请按照以下步骤解决：
> 
> ```
> sudo apt-get update
> sudo apt-get upgrade
> curl -sL https://deb.nodesource.com/setup_12.x | sudo bash -
> sudo apt-get install nodejs 
> npm install -g --unsafe-perm @frangoteam/fuxa
> sudo npm install -g --unsafe-perm @frangoteam/fuxa
> sudo fuxa start
> ```

Can you tell me which version of pi os you are using?

Brandix97, 2024-09-14:
I recommend to install with docker:

Install Docker:

# Add Docker's official GPG key:sudo apt-get updatesudo apt-get install
ca-certificates curlsudo install -m 0755 -d /etc/apt/keyringssudo curl
-fsSL https://download.docker.com/linux/raspbian/gpg -o
/etc/apt/keyrings/docker.ascsudo chmod a+r /etc/apt/keyrings/docker.asc#
Add the repository to Apt sources:echo \ "deb [arch=$(dpkg
--print-architecture) signed-by=/etc/apt/keyrings/docker.asc]
https://download.docker.com/linux/raspbian \ $(. /etc/os-release && echo
"$VERSION_CODENAME") stable" | \ sudo tee
/etc/apt/sources.list.d/docker.list > /dev/nullsudo apt-get update

install node red:

docker run -it -p 1880:1880 -v node_red_data:/data --name mynodered
nodered/node-red

Install nodejs

curl -fsSL https://deb.nodesource.com/setup_18.x | sudo -E bash -sudo
apt-get install -y nodejs

Install bcryptjs

sudo npm install -g bcryptjscreate pass
node -e "console.log(require('bcryptjs').hashSync('your_password', 8));"

Edit Settings file

sudo nano /var/lib/docker/volumes/node_red_data/_data/settings.js
Add the password (Important uncomment adminAuth)
adminAuth: { type: "credentials", users: [{ username: "admin", password:
"your_password_encrypted", permissions: "*" }]},

Restart

 docker restart mynodered