frangoteam/FUXA #1086: [BUG]Scripts doesn't execute well

Brandix97, 2024-02-18:
I have made a script that updates the date every second, and sometimes it runs every second and other times every 2 seconds

To Reproduce
Steps to reproduce the behavior:
1. Go to 'editor'
2. Create a script every second
3. Write this script:"var fechaActual = new Date();setag......"
4. put a text that the value is the date
5. Watch the error

**Environment**
 - Raspberry Pi
 - Node.js version:  14.21.3
 - Platform/OS: Raspbian x64
 - Browser: Opera GX

MatthewReed303, 2024-02-18:
I set my scripts only on start and then use the set interval JS function. 

```
//Dynamically grab tag ID
let tagNumber_1_ID = $getTagId('tagNumber_1');
let randomNumber = 0;
let inc = 1;
//Constant Vars
const minNum = 0;
const maxNum = 100;

let myVar = setInterval(funRandomNumber, 100);
function funRandomNumber() {
	randomNumber += inc;
	if(randomNumber == minNum || randomNumber == maxNum){inc = -inc;}
	$setTag(tagNumber_1_ID /* FuxaServer - tagNumber_1 */, randomNumber);
	//console.log(randomNumber);
}
```

unocelli, 2024-05-22:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-05-22 by unocelli