frangoteam/FUXA #1088: [BUG] Issues on panel control when using multiple clients

IonuDragomir1, 2024-02-19:
Hi,
I was testing FUXA's panel control on client 1 (C1) and was OK. After opening the same project on client 2 (C2) and starting to change screens called on panel, I noticed that it was reflecting also on C1's panel. This thing should not be happening. 
So if a project is used simultaneously  and C1 changes screens in a panel, should not be reflected on C2's panel control. 
**Example:** Imagine two or more engineers that have to follow a process but they can't do that because each one changes on a different screen inside of  same panel.

MatthewReed303, 2024-02-19:
@IonuDragomir1 I ran into the same issue and I have already passed this onto @unocelli and he is aware of the issue

IonuDragomir1, 2024-02-20:
@MatthewReed303 thanks and sorry. I didn't know that.

unocelli, 2024-02-20:
Hi, I suppose that you are using server tag to set the view, the server tag are available and the same for every client. Try to use internal device, that is only in frontend

IonuDragomir1, 2024-02-22:
Hi,
Using tags from internal device solves the problem. Thank you!

closed 2024-03-17 by unocelli