frangoteam/FUXA #1092: [Improvement] Could add jsonpath to parse complex json and get the depth node value?

ruijf, 2024-02-22:
FUXA is very cool!

When I use MQTT receiving json message, can only read top node value.
Could use jsonpath library to parse json to get depth node value? Just like this .
### Json message
```json
{
"dev_id": "d1001",
"dev_name": "dev-001",
"metrics":  {
    "mv1": true,
    "mv2" false,
    "mv3" false,
    "pump1": true
    }
}
```
### Define device tag
[image]
If can convert true/fasle to 1/0，will be more compatible. 

### jsonpath
https://jsonpath.com/

unocelli, 2024-03-02:
Hi, Thanks for the suggestions

Schrolli91, 2024-10-23:
I would highly appreciate this feature, to get easy access to nested nodes inside of the json messages.