frangoteam/FUXA #1095: [BUG] Icon not assigned to a button added on header bar

IonuDragomir1, 2024-02-26:
Hi,

I've just added a button on header bar from Layout. 
[image]
The button has assigned an icon which is not displayed on runtime. Name "Test" was assigned from button's property.

[image]

Environment:
FUXA version: 1.1.18 - 1431
Node.js version: v14.21.3
npm version: 6.14.18
Platform/OS: WIN 10
Browser: Chrome

unocelli, 2024-03-02:
Hi, the Icon is to use with Type: Image

andy8601, 2024-10-30:
In addition, the button added on the title bar does not have the user authorization option. I tried to set the authorization in the page, but it did not take effect, and users without authorization can still see and click to access the contents of the button after landing.

andy8601, 2024-10-30:
> 嗨，图标将与 Type： Image 一起使用

此外，在标题栏上添加的按钮没有用户授权选项。我尝试在页面中设置授权，但没有生效，未经授权的用户登陆后仍然可以看到并点击访问按钮的内容。

andy8601, 2024-10-30:
> 嗨，图标将与 Type： Image 一起使用

In addition, the button added on the title bar does not have the user authorization option. I tried to set the authorization in the page, but it did not take effect, and users without authorization can still see and click to access the contents of the button after landing.

mauroalexandre, 2024-11-06:
Pull request for fix [here](https://github.com/frangoteam/FUXA/pull/1485)

unocelli, 2024-11-10:
Thanks @mauroalexandre

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-12-11 by unocelli