frangoteam/FUXA #1098: [FEATURE]

henjoe, 2024-03-01:
**Describe the feature**
Object oriented graphic component.

**Describe the solution you'd like**
When developing an scada graphics, a specification design already given to us let say for example (pumps, valves and etc).
It would be nice if we have the capability to create this as graphic objects where we can add custom parameters (and we can link that to a Tag). 

**Additional context**
An example for this I woukd create  a graphics object for my pumps with the following customize parameters:
- status
- startCommand
- StopCommand
- Tag

And I can re-use this object for all of may views and simply edit the parameters. 
I know you get what I mean for this "Object Oriented approach". 
This features gives another level for this Fuxa SCADA.

IonuDragomir1, 2024-03-01:
Hi,
I suppose you mean to use a view with specified parameters. This view could be called from each pump and the parameters would get the values of a certain pump. Did I get it right?
So, for example, if you have 5 pumps, P1, P2...P5 and if would open the P3's window, you'd get P3's parameter values. If would open the P5's window, you'd get P5's parameter values, and so on.
Similar topics are #308 and #744  which are yet to be solved.

henjoe, 2024-03-01:
Hi thanks for your comment, howevere I believe its different concept, (they have similarities though).

Apologies to confuse, to make things simple, pets just say that the featured I want is 
I can create for example a view where I can drag and drop that view in another view.

Create View for Pumps with customized parameters/Properties like:
- status
- tagname
- and etc. 

Once that view is been use to another view, I can input new values of parameters on it (in design mode), so it create another instance of that view.

MatthewReed303, 2024-03-02:
Hi @henjoe I have spoke to @unocelli about something similar, basically saving a view and using it as symbol/widget on other views and can bind tags to the view and have scripting bound to that view for an animated widgets etc sound like you want to make a symbol/widget that can be saved and used through out views/projects. This comes up quite often from having dynamic SVGs etc 

From what I understand there are some limitations to the current version of the SVG editor Fuxa is using and not easy to update as it is no longer fully opensource/example available. I think if the user could add for example an SVG arrow and make it rotate around a point using JS and another SVG as static background and save that view as a widget you now have a dynamic symbol. This could be a work around in the mean time until elements of SVG can be modified and manipulated etc

henjoe, 2024-03-02:
Hi @MatthewReed303 thanks for this. Have you tried it already with success?

robbudge, 2024-03-13:
I think generally we are all looking for 're-usable User Widgets / Groups ' or some way to pass a variable into a group and have the tag linking done automatically at the group level, like utilizing $getTagId(var1+'.status') within the tag binding. Then declaring the variable var1 to the group.
Then as we copy and paste the group var1 becomes 'Motor1','Motor2' etc.
the graphic would then be Motor1.status or Motor2.status automatically
could $getTagId() be evaluated and replaced with the correct uuid at the client level ?

one short term help would be if 'Edit Bind of Tags' showed all tags if multiple objects or groups were selected.
currently it only shows a single binding

MatthewReed303, 2024-03-13:
@robbudge I have also noticed that edit bind of tags does not work with groups. I think if we had a way to export the group as a JSON, it could then be imported and used in other projects and shared with the community. We still need scripting within that group to provide logic for the widget. 

@unocelli what are your thoughts on fixing the edit bind of tags on grouped objects and then later making a way to have JS within the group or attached to group to provide logic for the widget? I think also having nested groups within a main group and the edit bind of tags can access the nested group.

closed 2024-10-27 by unocelli