frangoteam/FUXA #1099: [BUG] Bar chart doesn't work as expected

KamilGrabowski01, 2024-03-01:
**Describe the bug**
Regardless of the selected data range in the bar chart, it consistently displays data from the last 12 hours.

**To Reproduce**
1. Create a view containing a bar chart.
2. Choose any data range within the bar chart settings.
3. Observe that the chart consistently shows data from the past 12 hours, regardless of the selected range.

**Expected behavior**
When selecting a specific data range, such as the last hour, the bar chart should accurately reflect data from that selected time frame. This behavior should apply consistently across all data range options available.

**Screenshots**
[image]
[image]
[image]
[image]

**Environment**
- FUXA version: 1.1.17 - 1399
- Node.js version: 16.20.2
- npm version: 8.19.4
- Platform/OS: Windows 10
- Browser: Opera

Bundyra, 2024-03-01:
I observed the same problem.

unocelli, 2024-03-03:
Hi, what is your goal, show current Tag data (realtime) or the Tag history values typically used for KWh?
I tested and for me work as expected.
[image]

unocelli, 2024-03-17:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-03-17 by unocelli