frangoteam/FUXA #1101: [BUG] Euromap specific - ECONNREFUSED on injection press

SerialCoffee, 2024-03-04:
**Describe the bug**
When adding a connection to a KraussMaffei injection press, it seems that the OPCUA server on the machine side refuse the connection.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to connect to a KraussMaffei Injection molding press.

**Expected behavior**
I know for a fact that KM press are tedious when it comes to OPCUA...
You have to first connect to the machine **with the security already setup**, in order for the press to automatically refuse the certificate. 
Then, you have to physical go to the IMM (Injection Molding Machine) screen to manually validate the connection certificate.
This is very nice when it comes to security features, not so much when you have to set it up. 

My question is : 
Is there a way to set up the securities for the user before attempting to connect to the server? Else in my case, the connection cannot reach the server as there is a mean firewall in between.

Thanks for any informations related to that issue.

**Screenshots**
[image]

**Environment**
 - FUXA version: Latest (installed this morning) -> V1.1.17?
 - Node.js version: v20.11.0
 - npm version: 10.2.0
 - Platform/OS: Debian 11 (bullseye)
 - Browser: Firefox / Edge / Chrome

unocelli, 2024-03-12:
Hi, I don't know if this is the reason, but I would first try node.js version 18
You need to have installed Version 14.21.3 (npm 6.14.18) || 16.20.2 (npm 8.19.4) || 18.18.2 (npm 9.8.1).

SerialCoffee, 2024-03-13:
Hi, That is in fact something i didn't try.
I was wondering if that would be the case...

Digging on the subject, does FUXA works with asynchronous event from an OPCUA server (such as alarms and informations messages)?
On the meantime, i wil try to downgrade NodeJs according to what you said and run the app.

Thanks for coming back to me.

SerialCoffee, 2024-03-15:
Hi,
Coming back for more informations after my tries. I downgraded nodeJs to version 18, as asked, and the result is the same.

I do not know how the first access to an opcua server is made in fuxa, but i think i have an idea...
Does the fuxa server try to ask the opcua server first, and then ask for the securities? 

On this specific injection press brand, my tries with UAExpert told me that i have to trigger the first connexion with the id/password completed already, or else the firewall box between the server and the client refuses the access. 

Is there a way to fill the security before fuxa try to access the server?

Thanks,

tranquilo, 2024-03-15:
FUXA connects the same way as the node-opcua sample https://github.com/node-opcua/node-opcua-sample/blob/master/simple_client_es8.js
It first creates a connection and then the session. The credentials are passed after connection has been created, when the session is being created. At least is what it seems from the node.opcua library.
Now, what you might really want is the certificate support during the connection?
Checking the node-opcua client options, there are certificate related options that are currently not exposed by FUXA:
https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client/source/client_base.ts

tranquilo, 2024-03-16:
@SerialCoffee There is a workaround for adding the certificates, you can look into this issue that is related:
https://github.com/frangoteam/FUXA/issues/299

SerialCoffee, 2024-03-18:
Hello, thanks for bringing more insight on the problem, i will check that out and come back to you.

unocelli, 2025-10-03:
I assume the issue has been resolved in the meanwhile

closed 2025-10-03 by unocelli