frangoteam/FUXA #1104: [BUG] MODBUS TCP Doesn't Re-Establish after Link Loss

SFTP-ChZeman, 2024-03-07:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Verify you are receiving data from a MODBUS TCP device.
2. Disconnect Ethernet from the device and verify loss of link in FUXA.
3. Reconnect Ethernet and watch the Connections screen in FUXA. It will continually cycle between orange and green, but never re-establish communications.
4. Save the project as-is and communications will re-establish.

**Expected behavior**
I expected communications to re-establish and data begin updating on the screen again after the PLC was reconnected.

**Environment**
 - FUXA version: 1.1.17-1399
 - Node.js version: 18.19.1
 - npm version: 10.4.0
 - Platform/OS: Debian 12
 - Browser: Firefox 122.0.1 (64-bit)

robsori, 2024-03-09:
I confirm this problem on Windows OS as well.

unocelli, 2024-03-26:
Hi, Thanks for the report, we will try to solve the problem as soon as possible...any suggestion is welcome

unocelli, 2024-05-17:
Hi, a fix to test is available in master branch, would be nice to have feedback

unocelli, 2024-05-22:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-05-22 by unocelli