frangoteam/FUXA #1107: [BUG] Modbus RTU communication 4800bps time out before completion

harvie256, 2024-03-10:
When a Modbus RTU device is setup with a baud of 4800, FUXA does not wait long enough for the transmission to complete before throwing an error, e.g.:
"2024-03-10T04:59:22.271Z [ERR]      'Weather Station' _readMemory error! Error: Data length error, expected 7 got 1"

The device was tested with Qmodbus and python minimalmodbus library, and both worked as expected.

The serial traffic was sniffed, and verified all communication on the wire was happening as it should.

The issue was resolved by changing the device to use 9600 baud and updating FUXA.  FUXA now collects data as expected.  However this isn't a real fix, as lower baud rates are sometimes needed to communicate with devices at the end of long cable runs.

It looks like somewhere there is a timeout that is occurring after only the first byte is received.

Thanks

tranquilo, 2024-03-12:
When you configured the Modbus RTU device on FUXA, did you specify the other parameters besides `baudRate`?
Code reference:
```    var _connect = function(callback) {
        try {
            if (type === ModbusTypes.RTU) {
                const rtuOptions = {
                    baudRate: parseInt(data.property.baudrate),
                    dataBits: parseInt(data.property.databits),
                    stopBits: parseFloat(data.property.stopbits),
                    parity: data.property.parity.toLowerCase()                    
                }```

harvie256, 2024-03-12:
Yes, everything was set correctly.

External to FUXA, writing to the device to change baud rate to 9600, then changing in the baud rate in FUXA with NO other changes fixed the issue.

Where actually is the timeout for this transaction set?  I went hunting but couldn't find it (my knowledge of JS isn't great.) If I knew where the timeout was set I could modify and verify if that resolved the issue by changing the device back to 4800.

unocelli, 2024-03-26:
Hi, I'm not sure but could be a problem with the read delay. can you try to change the delay for example to 100 milli:
change fuxa\server\runtime\devices\modbus\index.js line 122 and 131 with `readVarsfnc.push(await delay(100));`

harvie256, 2024-03-28:
Thanks I'll give it try over the weekend!

dtx1234, 2024-05-08:
hi，Have you solved this problem?

unocelli, 2024-05-17:
Hi, I added the possibility to set a frame delay that maybe can help (in master branch), would be nice to try and give me feedback

harvie256, 2024-05-18:
Sorry I'm overseas for a few weeks, but will defiantly try it when I'm back.

xigua1688, 2024-07-15:
i also meet this issue.
changed fuxa\server\runtime\devices\modbus\index.js line 122 and 131 with readVarsfnc.push(await **delay(800));**

still have the errer logs as following,some times read the right value,some times get error  as "CRC error" or Data length error, expected 7 got 2: 

Read holding registers==============: { data: [ 2284 ], buffer: <Buffer 08 ec> }
Error: CRC error
    at ModbusRTU._onReceive (D:\code\fuxa\FUXA-master\server\node_modules\modbus-serial\index.js:416:14)      
    at SerialPort.emit (events.js:400:28)
    at addChunk (internal/streams/readable.js:290:12)
    at readableAddChunk (internal/streams/readable.js:265:9)
    at SerialPort.Readable.push (internal/streams/readable.js:204:10)
    at D:\code\fuxa\FUXA-master\server\node_modules\@serialport\stream\dist\index.js:208:18
2024-07-15T03:22:19.727Z [ERR]  'modbusRTU' _readMemory error! Error: CRC error
Read holding registers==============: { data: [ 2283 ], buffer: <Buffer 08 eb> }
Read holding registers==============: { data: [ 2282 ], buffer: <Buffer 08 ea> }
Read holding registers==============: { data: [ 2282 ], buffer: <Buffer 08 ea> }
Error: Data length error, expected 7 got 2
    at ModbusRTU._onReceive (D:\fuxa\FUXA-master\server\node_modules\modbus-serial\index.js:406:14)      
    at SerialPort.emit (events.js:400:28)
    at addChunk (internal/streams/readable.js:290:12)
    at readableAddChunk (internal/streams/readable.js:265:9)
    at SerialPort.Readable.push (internal/streams/readable.js:204:10)
    at D:\code\fuxa\FUXA-master\server\node_modules\@serialport\stream\dist\index.js:208:18
2024-07-15T03:22:27.743Z [ERR]  'modbusRTU' _readMemory error! Error: Data length error, expected 7 got 2
Read holding registers==============: { data: [ 2282 ], buffer: <Buffer 08 ea> }

this is logs get the right value from a modbusRTU device（usb--->485）
Read holding registers==============: { data: [ 2283 ], buffer: <Buffer 08 eb> }