frangoteam/FUXA #1109: [FEATURE]Could you add a feature for dynamically binding tag placeholders?

luy710, 2024-03-12:
This functionality would enable greater flexibility in reusing layouts where, for instance, within a page's tags we have references like A.{device}.value1, A.{device}.value2, and A.{device}.value3. By passing or switching the value of the device variable, we could achieve different configurations for various devices.

unocelli, 2024-03-17:
Hi, I agree, fuxa should better integrate the concept with placeholders

Jue07, 2024-05-02:
Hi, is this feature already available? if so, on what version is this placeholder available? what I found after installing Fuxa on several computers not all have this placeholder even though I have used the latest version of Fuxa which is v 1.1.18.

[image]

[image]

unocelli, 2024-05-07:
Hi, the placeholder feature is currently only in master branch available and will be released in v1.1.19 
You will be able to bind dynamically tags. not devices.

closed 2024-05-22 by unocelli