frangoteam/FUXA #1111: Hidden Button property does not get re-evaluated when the button is shown

robbudge, 2024-03-12:
Testing Beta 1.18.1445

Create 2 variables, Number Var1, Number Var2
On a dialog page create 3 buttons.
Button 1- Set Var 1 to 1 (Green if Var 1 =1)
Button 2- Set Var 1 to 2 (Green if Var 1 =2)
Button 3- Set Var 1 to 3 (Green if Var 1 =3)

Hide Buttons 2 & 3 if Var 2 > 0.
Set Var 2 to Zero.
Open dialog Click Button 2 (Var 1=2)
Set Var 2 to 1 (Button 2 is hidden)
Close Dialog.
Set Var 1 to 1, Var 2 to Zero.
Open Dialog
Button 1 & Button 2 will both be green.

unocelli, 2024-03-27:
Hi,
I have followed the instructions described but for me it works as expected, can you send me an example (project)?

unocelli, 2024-05-22:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.