frangoteam/FUXA #1113: [FEATURE] New Customizable Login Overlay Color Feature (#1112)

mauroalexandre, 2024-03-13:
We have introduced a new feature (#1112) that allows users to customize the overlay color on the login screen. This enhancement aims to improve user privacy and the overall user experience by enabling the obscuration of view content when users are not logged in. Additionally, it offers an option for users who prefer to maintain the default settings, ensuring flexibility and a non-disruptive integration into the existing platform.

## Purpose
- **Enhance User Privacy**: The customizable overlay color helps in keeping the underlying content of the application hidden from unauthorized access, thereby improving privacy.
- **Increase Customization**: Users can personalize the login screen to match their aesthetic preferences or the branding guidelines of the platform.
- **Optional Feature**: This feature is designed to be optional, catering to the diverse preferences of our user base.

## Request for Feedback
We are seeking feedback on the implementation, usability, and potential improvements of this new feature.

Thank you for your time!

unocelli, 2024-03-21:
Hi, thank you again for the feature, I just made a small change and in testing I noticed that if I cancel the login the first page is shown anyway. I was wondering if it is a good idea to hide the cancel button when you settings to show the login at start-up and background colour to white or black?

mauroalexandre, 2024-03-21:
> Hi, thank you again for the feature, I just made a small change and in testing I noticed that if I cancel the login the first page is shown anyway. I was wondering if it is a good idea to hide the cancel button when you settings to show the login at start-up and background colour to white or black?

Yay! That would be great, sometimes I'm afraid to suggest changes to the project to meet "my needs". Today, I have a docker compose, which makes some overrides in the FUXA files to accommodate some changes, which I'm "ashamed to propose".

One of them is to remove the cancel button from the login screen.

Look at the attached image as proof of my word. Hahahahaha

[image]

unocelli, 2024-03-22:
I will add the logic to hide the cancel button only if you enable login on start and background.
If you prefer you can contact me in discord so that we can discuss about your features :wink:

unocelli, 2024-05-22:
is available in the latest version.

closed 2024-05-22 by unocelli