frangoteam/FUXA #1115: [FEATURE]Default Security Level / User, View Only without Login

robbudge, 2024-03-14:
With Security and Users enabled.
The ability to specify the default user and therefor the security level would be a great feature.
it is often beneficial for general staff members to be able to view the system / status / alarms etc but not be able to make any changes.
Having the ability to specify the default / minimum user level to say viewer would allow the system to be viewable when no user is logged in, Or have the log out simple login a specified user say 'Viewer' with specific access rights.

It would appear currently that without a user logged in, even unrestricted graphics are not updated.

unocelli, 2024-03-26:
Hi, I am not sure what you mean, can you give an example or even a small project

robbudge, 2024-03-27:
This is more visual and screen size issue
by default our keyboard is numerical only. 
users only ever need to change setpoints and not enter text. 

Correctly on login the full keyboard is show. 

We typically use a 1920 x 1080 resolution on a 22” screen
This is about 500mm wide. 
Producing about 4 pixels per mm. 
the default keyboard / login size is only 250px
This results is the full qwerty keyboard being like 65mm wide on screen. 
The small full qwerty keyboard only being 66mm wide makes password entry difficult to hit the right keys. 

On my build I changed the width from 250px to 500px via the source files and rebuilt. 
The button sizes on our 22” screen works far better. 
I think as native resolutions increase the fixed 250px wide keyboard could cause some issues. 
perhaps a percentage option ?